revarbat/snappy-reprap #2: Documentation missing parts sizes

ArktekniK, 2015-09-16:
Hi, 

Having a look through the repo I can't seem to find details on the minimum bed size needed to print these parts?  

Assuming the STLs are in mm scale, it looks like the rail_segment_parts (for example) is too large to print on a Printrbot Simple Metal with a bed of 152*152*152mm.

andrewcaustralia, 2015-09-16:
The longest piece is the "rail_segment_parts.stl" which is 160.58 x 134.58 x 61.50mm
The tallest piece is the "z_joiner_parts.stl" which is 150.05 x 56.29 x 110.00mm

jkeegan, 2015-09-16:
Yeah unfortunately *my* Prusa IT2 (might be different for others based on extruder) can only print up to about 65mm tall before hitting the bars, which is the main reason I'm thinking of making a 3rd printer. The thing I dislike about the Prusa i3 is the frame - a large vitamin. The thing I dislike about the Taz is extruded aluminum rails - vitamins that you can't buy at your local hardware store or make easily in some 3rd-world country. This however is very tempting. Any idea *how many* of the objects are taller than 60-65mm? (I could see printing a few of the parts on someone else's printer and treat those first copies as vitamins and still count it as my RepRap printing it)..

ezrec, 2015-09-16:
I have a 180mm diameter Delta, but only printable to about 150mm diameter, so I'm going to have to scale-in-Y down a bit when I start printing these.

When I get to that point, I'll start a fork with SCAD files adjusted to the smaller size.

(Working on consistent bed adhesion at the moment and stringiness issues at the moment...)

ArktekniK, 2015-09-16:
I also thought about rescaling the parts but decided it is not a good idea because several parts interface/lock with motors, board mounts etc. Therefore I'm now at the point where I'll probably fork and modify the scad for the problematic parts.

@revarbat If I make suitable changes and submit a pull request would that be okay?

revarbat, 2015-09-17:
Here's the current part sizes.

part name                              | size
---------------------------------------|------------------------
cable_chain_link_parts.stl    | 142.0 x 137.0 x  13.0
cable_chain_mount_parts.stl  |   93.5 x  89.5 x  25.0
cooling_fan_shroud_parts.stl  |  84.0 x  44.1 x  19.1
drive_gear_parts.stl          |  47.9 x  48.0 x  20.0
extruder_fan_clip_parts.stl    | 70.0 x  48.0 x  12.3
extruder_fan_shroud_parts.stl |  50.0 x  48.0 x  39.7
extruder_idler_parts.stl      |  67.9 x  51.9 x  36.0
extruder_motor_clip_parts.stl  | 50.0 x  73.0 x  37.4
jhead_platform_parts.stl     |  120.0 x 140.5 x  60.0
motor_mount_plate_parts.stl   |  92.3 x  50.0 x  26.9
platform_support_parts.stl     | 70.0 x 123.0 x  43.0
rail_endcap_parts.stl        |   35.3 x 120.0 x  62.0
rail_motor_segment_parts.stl |  110.6 x 120.0 x  62.0
rail_segment_parts.stl       |  134.6 x 160.6 x  62.0
rambo_mount_parts.stl      |    120.0 x 165.0 x  55.8
ramps_mount_parts.stl       |   120.0 x 111.5 x  65.8
sled_endcap_parts.stl        |  157.7 x  23.8 x  40.0
sled_parts.stl               |  107.7 x 150.0 x  40.0
slop_calibrator_parts.stl    |   67.0 x 100.0 x  15.0
spool_holder_parts.stl       |  120.0 x 155.3 x  62.0
support_leg_parts.stl         |  60.0 x  88.2 x  50.0
xy_joiner_parts.stl           |  65.0 x 150.0 x  53.3
yz_joiner_parts.stl         |   142.6 x 116.6 x  67.3
z_joiner_parts.stl          |   150.1 x  56.3 x 110.0

Two parts exceed 65mm tall. The tallest part is 110mm.  The longest part is 165mm for the optional RAMBo mount, which can be improved.  The ubiquitous rail segment is 160.6mm long. (The motor mount plate was 177, but it was 2x2 repeat of the part, and I've committed a change to make it a single part.)

If 150x150 is the target build area, the rail_length can be reduced by 15, and the motor_rail_segment length can be increased by 130. I need to double check that that won't cause unfortunate side effects, but it should work.

revarbat, 2015-09-18:
Okay, I've pushed parts size changes so all part are printable in a 150x150mm area.  The only part longer than 150mm is the sled endocarp part.  It's hard to shrink, at 157.7mm, but it can be easily printed diagonally on a square build platform. Not sure how to shrink it below 150 without shrinking the platform width.

ezrec, 2015-09-18:
I'll try printing the 150mm sled end cap, and see if it'll fit on my 180mm delta.

ezrec, 2015-09-18:
Looks like the sled end cap and the yz_joiner_parts both fit! Nicely done.

(Though yz_joiner_parts was very close - I have to turn on off 'brim' and 'skirt', or I get a head crash when it tries to go so close to the edge.

andrewcaustralia, 2015-09-18:
Allowing them to fit in the 150mm bed diagonally is just as good. Do the changes require reprinting existing parts or have they just been modified to suit the current build (I ask as I've already got the STL's downloaded and am printing them out; am about 1/3rd of the way done).

[image]

ezrec, 2015-09-18:
Additional suggestions:

* Use just a single 'chain_link_part' STL, and let the builder do bed placement for the set.
*  jhead_platform_parts is too big by about 1cm to fit my delta.
* rail_motor_segment_parts is too big by ~1cm
* rail_segment_parts is too big by ~1cm
* sled_parts too big by ~1cm
* .. and the same with spool_holder_parts

All the rest fit (except the RAMBO holder, but I'm using RAMPS, so not my itch to scratch...).

So close!

I'll see if one of my Cartesian friends can print the problematic parts for me.

revarbat, 2015-09-18:
Unfortunately, the length of all rail segments have changed.  You CAN just set rail_length back to 150 and motor_rail_length back to 100 in config.scad, and regenerate your own parts.

I understand your pain, though. I just spent a week printing the parts needed to finally test the direct_drive_z changes, and I found out all the sled pieces have their slop set too tightly, so the sleds have way too much friction.

I'm afraid this is why this printer is still marked experimental.

The sled pieces haven't changed, so you will probably want to keep going with the STLs you have, then download the new sled, and reprint those.  Sadly you will need 8 of them.

If you want to try the last verified working version of the printer, you can go off of the v0.95 branch STLs, though they are for the lifter-rod based Z axis.

revarbat, 2015-09-18:
Ezrec, you can always regenerate the parts after tweaking with the rail_length and motor_rail_length in config.scad.  You can always print half-length rail segments (via rail_length) and use twice as many, for example.

andrewcaustralia, 2015-09-18:
Hey that's no worries mate; just figured I'd check and confirm! I'm sure I'll be reprinting parts for a while making some adjustments myself anyway ;-)

revarbat, 2015-09-18:
Ezrec you may also be able to get good results from reducing platform_width by 10 or so.  That will end up narrowing all the rail parts as well.

ezrec, 2015-09-18:
Thanks! I'll try that.

ArktekniK, 2015-09-19:
>You can always print half-length rail segments (via rail_length) and use twice as many, for example.
This is exactly what I had done - glad to see that you've suggested the same thing. Some quick maths lead me to 69.71 for the rail length. 75mm isn't right as it appears the connectors add about 10 mm to the total. Total of two of them together should be 160.58mm

ArktekniK, 2015-09-19:
>I just spent a week printing the parts needed to finally test the direct_drive_z changes, and I found out all the sled pieces have their slop set too tightly, so the sleds have way too much friction.  

Now that you mention it, I was just wondering what the slop calibration part is for? I assume we print it and the values are used in calibrating values for the config.scad?

revarbat, 2015-09-21:
You print a slop calibration part, to figure out the value to plug in to the printer_slop value in config.scad.  The hole that the plug fits into without a lot of effort, has the slop value printed on the side of the block in front of it.

The printer_slop value is used in a lot of places to enlarge or widen fittings where parts fit together.  This includes sliders, joiner tabs, mount holes, etc.

ezrec, 2015-09-21:
Thanks for pointing out the slot calibration part STLs.

Just printed it (55min print time), and got the '10' value as the best fit.

Machine: Geeetech Delta Rostock (modified), Repetier 0.91, E3D v6 Lite (0.8mm nozzle)

Media: PLA (FoxSmart) 1.75mm, 0.7mm layer height, 30% infill, 44mm/s, 210C, no heated bed.

Looks like I should be able to churn these out once I get my new spools.

jkeegan, 2015-09-21:
Brilliant!

revarbat, 2015-09-27:
Okay, just a heads up... I have repeatedly failed to get the direct drive Z axis mods to work, so I'm reverting back to using the two lifter rod design for the Z axis.  This means, for those already having printed a lot of parts, you'll need to print the following new files:
- z_sled_parts.stl
- rail_z_motor_segment_parts.stl
- lifter_rod_coupler_parts.stl
- lifter_lock_nut_parts.stl
- rail_endcap_parts.stl

These replace the follow parts:
- 4 of the 8 sled_parts.stl
- 2 of 4 sled_endcap_parts.stl
- 2 of 4 rail_motor_segment_parts.stl
- Both z_joiner_parts.stl
- 1 of 3 rail_endcap_parts.stl

ArktekniK, 2015-09-28:
Thanks for the info - I just finished printing the XY joiner parts when I saw the new check in for 1.1b. Did a quick comparison and saw it was updated. The two vertical parts with the connectors on top seem to have moved inwards a few millimetres. Will they need to be re-printed too?

revarbat, 2015-09-28:
Yeah, the XY joiners were incorrect due to the earlier parts resizes, so they had to be fixed. You probably need to print one new rail_xy_motor_segment_parts.stl part as well, to get one with side joiners.

closed 2015-10-04 by revarbat

3dartsinc, 2016-01-23:
I did not find any download how can i download snappy stl. files