revarbat/snappy-reprap #3: YZ Joiner cable hole problem with crossbeam style walls

ArktekniK, 2015-09-26:
Hi,
I am experiencing lots of shrinkage with the default STLs, so I switch the wall styling and noticed YZ joiner has a problem with crossbeam style walls. The hole is too large and causes the beams not to meet. TO be honest, I don't think the hole is needed if you've chosen crossbeam walls.

Lines 98-100 of YZ joiner parts should be wrapped in a conditional
https://github.com/revarbat/snappy-reprap/blob/master/yz_joiner_parts.scad#L98#L100

Something like:
```
// Wiring access hole
if (wall_style != "crossbeams") {
    up(rail_thick+rail_height/4/2) {
      cube([16, joiner_width*3, rail_height/4], center=true);
    }
}
```

ArktekniK, 2015-09-26:
[image]

revarbat, 2015-09-27:
Good catch.

revarbat, 2015-09-27:
[v1.1 0082da0] Fixed issue #3: YZ Joiner cable hole problem with crossbeam style walls
 1 file changed, 4 insertions(+), 2 deletions(-)

closed 2015-09-27 by revarbat