revarbat/snappy-reprap #6: YZ Joiner cable hole problem with corrugated style walls

ArktekniK, 2015-10-15:
Hi there, just thought I'd mention that I noticed when this object is sliced without support, the printer will attempt to print unsupported overhangs. This is because the bridge is not a direct line across the gap.
 

[image]

revarbat, 2015-10-15:
Oh dear.  I  never really print with the corrugated walls, so I didn't see this.  There will be a number of places with this issue, I think.  It should be fixable by making a frame around the wiring access holes, though.

revarbat, 2015-10-15:
[master 0e05308] Fixed issue #6: Wiring access holes need frames on corrugated walls.
 4 files changed, 42 insertions(+)

closed 2015-10-15 by revarbat