revarbat/snappy-reprap #7: Scale model?

jpvlsmv, 2016-11-07:
Would it be possible to virtually build the printer at a size that can be printed out?  It would make a neat desktop toy.

Is the animated graphic rendered from a design file somewhere?

revarbat, 2016-11-26:
Huh.  Should be doable?  The renderings are generated from the full_assembly.scad file.  Should be able to load that file in OpenSCAD, then use F6 to render it with CGAL, which will take half an hour or so, then export it as an STL.  Not sure if it'll confuse the slicer with coincident faces in the joiners, but it'll probably work, I think.

revarbat, 2017-04-03:
The animations and renderings are generated from the full_assembly.scad file.

closed 2017-04-03 by revarbat