revarbat/snappy-reprap #8: J-head compatible

farfadet46, 2017-03-21:
Hi, i found your project very intersting but i only this model of Jhead (http://www.threedeeprints.com/product/3d-cam-metal-j-head-v6-hot-end-for-reprap-3d-printer-1-75mm-filament-direct-feed-extruder-0-4mm-nozzle/)
So, can you explain me how to put it in the extruder bridge ?
i have already printed it but it don't stay in.
https://github.com/revarbat/snappy-reprap/blob/master/docs/assembly/extruder_bridge_assembly_4_after.png

regards,

popotropopic3, 2017-03-21:
> head-v6-hot-end-for-reprap-3d-printer-1-75mm-filament-
> direct-feed-extruder-0-4mm-nozzle/)
> docs/assembly/extruder_bridge_assembly_4_after.png

I used a dremel. I would be happy to chat.

farfadet46, 2017-03-22:
A dremel oO, no no no, the j-head are too small to fit and stay in.
i think 'maybe' i need this part ?
https://sc01.alicdn.com/kf/HTB1rm4_GXXXXXb2XVXXq6xXFXXXO/200663255/HTB1rm4_GXXXXXb2XVXXq6xXFXXXO.jpg

Sorry for my english, i'm French, so, some time i don't use the correct grammar :p

popotropopic3, 2017-03-23:
> 200663255/HTB1rm4_GXXXXXb2XVXXq6xXFXXXO.jpg

My French is far worse than your English.  haha. Are you using a bowden
setup?

farfadet46, 2017-03-23:
hi ! no i think bowden isn't interesting for your printer

i think the thick of the platform is smaller than the J-head

https://sc01.alicdn.com/kf/HTB1rm4_GXXXXXb2XVXXq6xXFXXXO/200663255/HTB1rm4_GXXXXXb2XVXXq6xXFXXXO.jpg
on this picture you can see 4.6 mm my J-head is about 5.9mm

the platform is about 4.5mm.
so maybe you can tell me what i can do to fix the j6head to the platform ?
i see on this https://github.com/revarbat/snappy-reprap/blob/master/docs/assembly/extruder_bridge_assembly_8_after.png maybe the component for fixing the ventilator can fix the j-head in place too ?

I just printed the platform for now ; about 6 hours it's my biggest time of print.
and 2 legs haha :)

best regards

popotropopic3, 2017-03-23:
> ssembly/extruder_bridge_assembly_8_after.png maybe the component for

I just used a few layers of Kapton tape. That helped make it a tight fit.

farfadet46, 2017-03-23:
i need 1mm of kapton tape XD ok i print the "extruder_fan_shroud_parts.stl"
wait and see ;)
big thanks !

farfadet46, 2017-03-25:
Hello, i just made an adaptator i'm surprised about how it's easy to make :)
[image]

if you want to add to your files (i named it 13... i don't know why XD)
[13.zip](https://github.com/revarbat/snappy-reprap/files/870382/13.zip)

revarbat, 2017-03-30:
The E3Dv6 head does not conform to the J-head standard. It conforms to the E3D mount standard.  :) It has different mount dimensions.

I worry that the all-metal E3Dv6 isn't workable anyways. I suspect it will carry too much heat up to its mount point and warp the PLA there. As it is, a J-head with its PEEK barrel gets hot enough to warp the PLA mount point if you don't have the cooling fan running on it. Let me know how your adaptor works for you.

popotropopic3, 2017-03-30:
I figured that out the hard way...

closed 2017-04-03 by revarbat