revarbat/snappy-reprap #9: ACME threaded rod - metric unitrs

NixonMroq, 2017-04-02:
Could someone please add link or spreadsheet to compatible ACME threaded rod in metric unit?
It is almost impossible to get one of these from BOM list in Europe.

revarbat, 2017-04-03:
I'm actually in the middle of building and testing Snappy v1.5, which does away with the threaded rods completely.

If you still want to work with v1.1, though, you can alter the lifter_rod_pitch value in config.scad from 3.175 to something like 2mm, and regenerate the z_sled_parts.scad and  lifter_lock_nut.scad models.

Just load each scad file into OpenSCAD, render with the F6 key, then export the rendered model to STL from the file menu.

revarbat, 2017-04-03:
You'll probably need to alter the lifter_rod_diam and lifter_rod_length values as well.

closed 2017-04-03 by revarbat