revarbat/snappy-reprap #10: Extruder quick-fit?

qwertystop, 2017-04-27:
I just found [this design](http://www.thingiverse.com/thing:19590) for a plate that allows easy swapping of extruders. As-is, though, that design only works for printers that mount the extruder on a bar. Is there any chance of a compatible equivalent for the Snappy? It would be useful to allow the use of e.g. [pellet extruders](https://www.youmagine.com/designs/universal-pellet-extruder-reprap-3d-printing).

revarbat, 2017-04-27:
Hmm. I may need to print one out to work out a compatible geometry.  The Snappy v1.5 changes may make it more challenging as well.

- Revar

ManuGithubSteam, 2018-12-18:
The new 3.1 design is quite quick fit, just undo the extruder and unplug it. It depends how you do the cables but my guess is that you can swap it in 3 -5 minutes

revarbat, 2019-02-23:
If someone else wants to implement that specific quick change mount, I can include it, but I've decided not to work on this myself.

closed 2019-02-23 by revarbat