revarbat/snappy-reprap #11: Amazon Links lead to unavailable items or confusing items

blade2005, 2017-05-09:
On BOM in the wiki, 
1. RAMPS kit is out of stock
2. Power supply, the link doesn't have a 120W or 300W option.
3. Cooling fan is out of stock, https://www.amazon.com/dp/B00E1JO8SG should replace it
4. Speaker wire is unavailable. i couldn't find something suitable 22GA thats as cheap.
5. The heated bed is also unavailable, though I found https://www.amazon.com/dp/B00QQ5Q3BI/ which might be substituted.

revarbat, 2017-05-12:
If this in reference to the v1.1 or the v1.5 Wiki BOM?

blade2005, 2017-05-16:
I just saw this response. Sorry about the delay. I got to this from reprap.org
http://reprap.org/wiki/Snappy which at the bottom led me to this 
https://github.com/revarbat/snappy-reprap/wiki/BOM

revarbat, 2017-05-16:
Okay, that's the v1.1 links then.

blade2005, 2017-05-17:
It looks like all the links have been updated. Thanks :D

closed 2017-05-17 by blade2005