revarbat/snappy-reprap #14: Z Axis Issue

franzflasch, 2017-07-10:
Hi! 

I just built a snappy 2.0. I think the Z axis needs some redesign. How is this even supposed to work? The z sled parts always get jammed into the z-rail parts, as I think they are just to tight together. Have you ever built and tested the current version of the z-axis design?

I am not sure if it's really a good idea to replace the metal lifter screw rods with printed worm gears. 
Any advice to this issue?

revarbat, 2017-07-10:
Does the Z binding go away if you pry the YZ joiner segment a couple mm away from the X axis rail on one side?  I'm starting to think I may have the bridge segments very slightly too long.

franzflasch, 2017-07-10:
Hi revarbat! Thanks for your answer. 

I think the problem is that the space between the motor-case of the z-sled part and the z-rail part is just too tight. The bridge segment distance looks OK to me, but I can test that.

But honestly I am thinking of just use the z-axis design from the snappy 1.1 and combine it with the new bridge. I would just print the old z-axis towers and adapt the z-axis-bridge joiner part to the new angle of snappy 2.0.

Do you see any problem in this approach?

franzflasch, 2017-07-12:
Here's a picture, maybe now it's easier for you to see the problem:

[image]

As you can see I needed to break away some material of the z-rail parts in order to make it work. Otherwise the motor gets stuck.

revarbat, 2017-11-11:
This should be fixed in commit 0370cf7.

closed 2017-11-11 by revarbat