revarbat/snappy-reprap #15: I built one! Some issues to clear up

mw0uzo, 2017-07-17:
Hi there. Great work on the design of Snappy II :)
I have built mine in ABS and made several modifications to make it more workable.

1) Filed off plastic on the Z axis that catches on the Z sled. I have not made any modifications to the code, as the issue was found after printing 4 of the parts. Increasing the amount cut out in the design or cutting out a different shape should fix this.

2) Modified config.scad for increased build plate height to accommodate a heated bed. The bed supports printed in ABS work fine at 90 degrees and above. I have been using 100 degrees for several long prints without any sagging or warping of the support plastic, as the corners are significantly cooler than the centre.

config.scad
```
// Standard 200mm square borosilicate glass build platform
glass_width     = 216;   // mm
glass_length    = 214;   // mm
glass_thick     =   4.6;   // mm  3mm glass and 1.6mm bed
```

3) Modified config.scad and cable_chain.scad for a much better chain. The original design and values did not work well for me, they did not hold together strongly, tended to come apart, were stiff and needed some material filed off to work at all. The modifed part and change in dimensions work a zillion times better.

config.scad
```
// Cable chain dimensions
cable_chain_height = 15;  // mm
cable_chain_width  = 25;  // mm
cable_chain_length = 26;  // 26mm
cable_chain_pivot  =  5;  // 6mm
cable_chain_bump   =  1.5;  // mm
cable_chain_wall   =  3.25;  // mm
```

cable_chain.scad
```
include <config.scad>
use <GDMUtils.scad>

module cable_chain_barrel()
{
	h = cable_chain_height;
	w = cable_chain_width;
	l = cable_chain_length;
	r = cable_chain_pivot/2;

	color("SpringGreen")
	union () {
		difference() {
			up(h/2) cube(size=[w, l-20, h], center=true);
			up(h/2-2/2) chamfcube(size=[w-4*cable_chain_wall, l, h-cable_chain_wall-2], chamfer=2, center=true);
			zrot(40) cube(size=[4, (l-15)/cos(40), 2*cable_chain_wall], center=true);
		}
	}
}
//!cable_chain_barrel();

module cable_chain_mount1()
{
	h = cable_chain_height;
	w = cable_chain_width;
	l = cable_chain_length+2;
	r = cable_chain_pivot/2;

	color("SpringGreen")
	union () {
		difference() {
			// Sides and tabs
			mirror_copy([1,0,0]) {
				translate([(w/2-3*cable_chain_wall/2-printer_slop/2)-0.25, (-l/4)+0.5, h/2]) {
					cube(size=[cable_chain_wall-printer_slop, l/2, h-0], center=true);
				}
			}

            // Swap around these shapes, it works better the other way around!
            
            
			// Chamfer Bottom Front
			//translate([0, -l/2, 0]) {
			//	scale([1, tan(30), 1]) xrot(45) cube(size=[w+2, (h/2*sqrt(2))+0, h/2*sqrt(2)], center=true);
			//}

			// Round out top front.
			translate([0, -l/2, h]) {
				difference() {
					cube(size=[w+2, h, h], center=true);
					translate([0, h/2, -h/2]) {
						yrot(90) cylinder(h=w+3, r=(h/2)-0.5, center=true, $fn=32);
					}
				}
			}
        
        

			// Pivot Divot
			mirror_copy([1,0,0]) {
				translate([(w-cable_chain_wall)/2, 0, h/2]) {
					translate([(-cable_chain_wall/2-cable_chain_bump/2), (-l/2+r+2.75-printer_slop)+1, 0]) {
						yrot(90) cylinder(h=cable_chain_bump+2, r1=r, r2=r+cable_chain_bump, center=true, $fn=32);
					}
				}
			}
		}
	}
}
//!cable_chain_mount1();

module cable_chain_mount2()
{
	h = cable_chain_height;
	w = cable_chain_width;
	l = cable_chain_length+2;
	r = cable_chain_pivot/2;

	color("SpringGreen")
	union () {
		difference() {
			// Sides and tabs
			mirror_copy([1,0,0]) {
				translate([(w-cable_chain_wall)/2, 0, h/2]) {
					translate([0, l/4, 0])
						cube(size=[cable_chain_wall, l/2, h], center=true);

					translate([0, l/2-h/3, 0])
						yrot(90) cylinder(h=cable_chain_wall, r=cable_chain_pivot/2+1.333, center=true, $fn=64);

					// Pivot bump
					translate([-cable_chain_wall/2-cable_chain_bump/2, (l/2-h/3)+0.5, 0])
						yrot(90) cylinder(h=cable_chain_bump, r1=r, r2=r+cable_chain_bump-0.5, center=true, $fn=32);
				}
			}

			// Chamfer top back
			translate([0, (l/2), h]) {
				for (ang = [5:10:45]) {
					scale([1, sin(ang), cos(ang)]) xrot(45) cube(size=[w+2, h/2*sqrt(2), h/2*sqrt(2)], center=true);
				}
			}
		}
	}
}
//!cable_chain_mount2();

// vim: noexpandtab tabstop=4 shiftwidth=4 softtabstop=4 nowrap
```

4) Snappy II absolutely needs two Z endstop switches on each side. Setting it up is a real pain otherwise. I modified the Z endcap so that the spool holder could bit fitted on top, so two switches can be used and two spool holders.

rail_z_endcap_parts.scad
```
include <config.scad>
use <GDMUtils.scad>
use <joiners.scad>

module rail_z_endcap()
{
	joiner_length=40;
	w = z_joiner_spacing + joiner_width;

	color("YellowGreen")
	prerender(convexity=10)
	union() {
		difference() {
			union() {
				// Bottom.
				translate([0, -joiner_length/2, rail_thick/2])
					cube(size=[w, joiner_length, rail_thick], center=true);

				// Back.
				translate([0, -joiner_length+rail_thick/2, rail_height/2]) zrot(90) {
					if (wall_style == "crossbeams")
						sparse_strut(h=rail_height, l=w-0.1, thick=rail_thick, strut=5);
					if (wall_style == "thinwall")
						thinning_wall(h=rail_height, l=w-0.1, thick=rail_thick, strut=5);
					if (wall_style == "corrugated")
						corrugated_wall(h=rail_height, l=w-0.1, thick=rail_thick, strut=5);
                }

				// Endstop clip
				fwd((endstop_depth+2)/2) {
					left((z_joiner_spacing-joiner_width-endstop_thick-2+0.05)/2-5) {
						up(rail_height-(endstop_length+2*2)/2) {
							difference() {
								left(5/2) cube([endstop_thick+2+5, endstop_depth+2, endstop_length+2*2], center=true);
								left(2/2) back(2/2) {
									cube([endstop_thick+2*printer_slop+0.05, endstop_depth+0.05, endstop_length+2*printer_slop], center=true);
									cube([endstop_thick+2*printer_slop-2, endstop_depth+10, endstop_length+2*printer_slop-1], center=true);
								}
							}
							down(endstop_length/2+2+endstop_thick*2/2+5/2-0.05) {
								left((endstop_thick+2)/2+5) {
									right_half() trapezoid([0.05, 0.05], [2*(endstop_thick+2+5), endstop_depth+2], h=endstop_thick*2+5, center=true);
								}
							}
							zspread(endstop_hole_spacing) {
								right(endstop_thick/2-1.5/2+0.05) {
									back(endstop_depth/2+2/2-endstop_hole_inset) {
										scale([0.5, 1, 1]) {
											sphere(d=endstop_screw_size, center=true, $fn=8);
										}
									}
								}
							}
						}
					}
				}
			}

			// Clear space for joiners.
			up(rail_height/2-0.05) {
				joiner_pair_clear(spacing=z_joiner_spacing, h=rail_height, w=joiner_width, a=joiner_angle, clearance=1);
			}
		}

		// Joiner clips.
		up(rail_height/2) {
			xspread(z_joiner_spacing) {
				
                fwd(0) joiner(h=rail_height, w=joiner_width, l=joiner_length-5/2, a=joiner_angle);
                fwd(rail_height-4) zrot(180) yrot(180) joiner(h=rail_height, w=joiner_width, l=13, a=joiner_angle);
                
                
                //fwd(l/2) zrot(180) xspread(z_joiner_spacing) joiner(h=rail_height, w=joiner_width, l=10, a=joiner_angle);
					//back(l/2) xspread(z_joiner_spacing) yrot(180) joiner(h=rail_height, w=joiner_width, l=10, a=joiner_angle);
			}
		}
        
        
        // extra Joiner clip.
        //up(rail_height/2) {
         //   xspread(z_joiner_spacing) {
         //       joiner(h=rail_height, w=joiner_width, l=6+joiner_length/2, a=joiner_angle);
         //   }
        //}
	}
}
//!rail_z_endcap();

module rail_z_endcap_parts() { // make me
	zrot(-90) rail_z_endcap();
}

rail_z_endcap_parts();

// vim: noexpandtab tabstop=4 shiftwidth=4 softtabstop=4 nowrap

```

5) The positional accuracy of the Z axis was not good to start with. I improved mine by writing some gcode to exercise the Z axis up and down with no lubricant, so the lifter screw and z axis thread wore into each other, until a visible wear line was present on the bottom half of each thread. Then I lubricated the screws and threads with Vaseline. The improvements were significant, but there is still a little wobble along the z axis. I think you can obtain good accuracy with just a few more modifications.
- Rails on the Z axis. It looks like you had rails, but then got rid of them. I wonder why this was? Perhaps the current design just needs a rectangular channel down the Z axis with a piece poking out of the bridge motor pieces to prevent any X movement.
- A snap together bar that fits from one side of the z rails to the other, e.g. on the top of the z switch caps or on the spool holders if two are fitted. This would stop any movement along the x axis of the z rails and improve stiffness of the z axis design. Also, the filament could feed through the centre, stopping it from tugging on one side when consuming filament.

I'm very pleased with how this works for the first test version on the V2 design. I think you *can* get that Z axis accuracy without the threaded bars!

mw0uzo, 2017-07-17:
6. The LCD display cover did not seem to fit correctly with me, the clips were too thin and prone to break and the KILL button was not cut out. (This could have been my slicer). 2mm extra bezel depth and a smaller internal lip would make it fit together a lot better.

Pics of the machine
[image]
[image]
[image]

mw0uzo, 2017-07-17:
Some boxes printed out in PLA. The big case half is the entire size of the bed!
[image]

Here is a close up of the wobble on the Z axis. As you can see, it's actually pretty good considering there is zero threaded bar in the design! If this can be further improved, V2 will be a success imo.
[image]

mw0uzo, 2017-07-17:
I have one unresolved issue:

- Circles not printing well. They are low resolution and have flat sides. While they are generally circular, the flat sides make reproducting the lifter screws impossible at the moment, they are not good enough to work well. I think this might be the Marlin firmware. Any suggestions would be welcome on how to improve this.

mw0uzo, 2017-07-17:
Here are some settings in Marlin for acceleration etc to get trouble free prints:

#define DEFAULT_AXIS_STEPS_PER_UNIT   { 40, 40, 400, 96.0 }
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 10, 30 }
#define DEFAULT_MAX_ACCELERATION      { 1000, 1000, 10, 10000 }
#define DEFAULT_ACCELERATION          400    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  1000   // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   600    // X, Y, Z acceleration for travel (non printing) moves
#define DEFAULT_XJERK                  5
#define DEFAULT_YJERK                  5
#define DEFAULT_ZJERK                  0.4
#define DEFAULT_EJERK                  3
#define INVERT_X_DIR true
#define INVERT_Y_DIR false
#define INVERT_Z_DIR true
#define INVERT_E0_DIR true
#define INVERT_E1_DIR false
#define INVERT_E2_DIR false
#define INVERT_E3_DIR false
#define INVERT_E4_DIR false

I slackened off the monitoring of bed temperatues a lot. Otherwise too many false bed heating errors etc.

configuration_adv.h
#define THERMAL_PROTECTION_PERIOD 60        // Seconds
#define THERMAL_PROTECTION_HYSTERESIS 20     // Degrees Celsius
#define WATCH_BED_TEMP_PERIOD 120                // Seconds
 #define WATCH_BED_TEMP_INCREASE 2               // Degrees Celsius

For each print, the following G code must be entered into the slicer. This makes sure the right accelerations are chosen etc, so that the X axis never skips. I settled on using Repetier-Host as it was the most reliable at slicing, no crashes and coped with a lot of different designs. I tried others, Slicer etc but they were not reliable enough.

Start g code
; Default start code
G28 ; Home extruder
G1 Z15 F{Z_TRAVEL_SPEED}
M107 ; Turn off fan
G90 ; Absolute positioning
M82 ; Extruder in absolute mode
{IF_BED}M190 S{BED}
; Activate all used extruder
{IF_EXT0}M104 T0 S{TEMP0}
G92 E0 ; Reset extruder position
; Wait for all used extruders to reach temperature
{IF_EXT0}M109 T0 S{TEMP0}
; Set accelerations for Snappy II capabilities
M201 X1000 Y1000
M202 X1000 Y1000
M204 X1000 Y1000

End g code
; Default end code
;G1 X0 Y0 Z130 ; Get extruder out of way. Uncomment to use!
M107 ; Turn off fan
; Disable all extruder
G91 ; Relative positioning
{IF_EXT0}T0
{IF_EXT0}G1 E-1 ; Reduce filament pressure
M104 T0 S0
G90 ; Absolute positioning
G92 E0 ; Reset extruder position
M140 S0 ; Disable heated bed
G0 X96 Y194 Z167 F1000 ; Present to front
M84 ; Turn steppers off

Here are the defined speeds for the Cura slicer. Found by lots of trial and error
[image]

mw0uzo, 2017-07-17:
Another issue I encountered was Z axis skipping. This was resolved by running the other Z axis motor with the spare driver and enabling dual motors in the firmware. (Also dual switch Z axis homing was enabled)

mw0uzo, 2017-07-17:
One more note: For heating the bed to ABS temperatures, a heatsink must be fitted to the bed heater control MOSFET on RAMPS, or it *will* overheat and die. Temperatures over 105-110 degrees will be difficult to attain without increasing supply voltage, or upgrading the MOSFET as there is a reasonable amount of voltage drop across the original part and wiring leading to the bed.

mw0uzo, 2017-07-20:
Circles problem: fixed.
X compensation: 0.33
Y compensation: 0.1

Old lifter screw not usable - Marlin 0.3mm layer size - significant flat edge, egg shaped.
[image]
Usable lifter screw - Repetier 0.2mm layer size, no flat edge of any significance, circular.
[image]

As there is no compensation in Marlin, Repetier Firmware is the way to go. Compensation is necessary for Snappy II as play on my machine is minimal and will change over the lifetime of the gears and sliders.

Here is a zip of the firmware, preconfigured for Snappy II with Dual Z motors, 2nd Z motor connected to E1/Spare driver and dual Z endstop switches, the extra one connected to ZMin in the usual fashion.

[Repetier-Firmware-SnappyII-MW0UZO-RepRapRAMPS1-4-July2017.zip](https://github.com/revarbat/snappy-reprap/files/1162703/Repetier-Firmware-SnappyII-MW0UZO-RepRapRAMPS1-4-July2017.zip)

mw0uzo, 2017-07-24:
updated firmware - PID settings for extruder were way off. Set for J head extruder
[Repetier-Firmware.zip](https://github.com/revarbat/snappy-reprap/files/1171126/Repetier-Firmware.zip)

I have modified the spool holders to fit a bar across the top of the machine. Printing out the new parts to check them, then I'll post them here.

mw0uzo, 2017-08-02:
Modified spool holder and strut. Print 3 pieces for the strut.
[modified_pool_holder_and_xbar.zip](https://github.com/revarbat/snappy-reprap/files/1193617/modified_pool_holder_and_xbar.zip)
[image]

xyzprints, 2017-09-01:
wow! that's some work out there I like the Idea for the struts for the z axis!
btw, did you have some problems with the z axis wobble? I am having some serious wobbling:( do you got anny idea for this?

xyzprints, 2017-09-05:
(never mind that was a dumb question) also, what's back lashing and what kind of oil do you use for oiling the gears Thanks for the firmware!

mw0uzo, 2017-09-06:
Yeah mine was really bad until I exercised the Z axis up and down for ages. Here are some gcode files to do that:
[snappy-wear-in.zip](https://github.com/revarbat/snappy-reprap/files/1280459/snappy-wear-in.zip)

printerexerciseallaxis.gcode - do this for wearing in the XY bed after building. Ensure XY bed is level as possible when building. Superglue the Y axis segments to prevent sag. With the motors removed, check for how free the sliding parts are. Initially they will be quite stiff and some material will need to be filed off the sleds. Take a small amount off each time. Don't take off too much material from the sleds beforehand - you don't want any play. It's better to have some interference and resistance which will wear and free up as the machine is used. The machine will probably jump steps at first, turn up the motor current until jumping stops and ensure you have fan cooling on the electronics.

printerfindXYcurrent.gcode - When the machine is worn in, you can readjust motor current to get smoother prints. Run this and adjust current on the XY drivers to get smooth movement, no judders and no skipping. I run the Z axis motors and extruder at high currents. 

printerexercisezonly.gcode - This runs the Z axis up and down. Clean off any lubricant with a rag and IPA. Then run this continously to wear in the Z gears and rack. Turn up the current on the Z axis. Make sure you're running each Z motor on separate drivers, or there will not be enough current. I ran mine until i saw a wear line on the bottom half of the Z rack threads. This improved Z axis wobble a lot, but it is still present to some extent, a design change is needed to remedy it.

The strut removed a high frequency element of the Z axis wobble.
The Z axis wear reduced the amplitude of the wobble.
There is a low frequency element that I have not been able to affect so far. I think this requires the Z axis ends to be fixed in XY position on rails.

mw0uzo, 2017-09-06:
For lubricant, first I used 3 in 1 oil. But this is too thin and soaks into the plastic. Now I used Vaseline which doesn't need applying all the time and seems to work great. The only downside is when the machine has been standing for a while, the XY sled sticks initially, but once the Vaseline breaks free it works fine. Maybe I just put too much on.

xyzprints, 2017-09-06:
Whoa, that is some intense work out there! Thanks for the .Gcode! (is this .gcode for the marlin firmware?) I'll try to get this z axis to wear up. Also, I thought about using silicon oil for lubricant. would that be a good idea?

mw0uzo, 2017-09-06:
Yeah that would probably work. The problem with thin lubricants is that they soak into the grooves of the plastic and quickly drain away, so you'll go through a bottle or a can fairly quickly and end up with a puddle underneath when you move the machine. Use Vaseline, its cheap and works.

mw0uzo, 2017-09-06:
The gcode will work with Marlin or Repetier. The Z axis will move faster with Marlin. I set a hard limit for the Z axis speed in Repetier. When running the gcode at first with Marlin, watch the axis carefully and be prepared to yank the power if the Z axis gets jammed on one side. It can get wonky and the motors are powerful enough to snap or bend plastic on the bridge.

revarbat, 2017-11-17:
Okay, in order:

1. I've fixed the clearance on the Z rails, and modified the corners of the Z sleds.

2. I've added the thicker build plate value as a comment in config.scad.

3. Cable chains have been updated to use the values and tweaks you gave, more or less.

4. The Z end-caps have been modified so the spool holder can mount on top, and both towers can have a limit switch.  Also, I added a tower-to-tower top brace to help with Z wobble and stalling.

5. The Z rails and sleds used to have rail sliders, but I had great problems with the Z axis stalling and getting stuck.  I was using a single Z motor driver on the RAMPS, though.  I may have to try again with dual drivers.

revarbat, 2017-11-17:
Regarding circles not printing well, that is indeed a problem with backlash. The press-fit drive gears have too much play in them when they change directions.

My solution is to print the drive gears with extra-high infill density, then slide a M3 nut into the slot on the bottom, then thread in a M3 screw (10 or 12mm long) through the nut, then tighten it against the flattened shaft of the drive stepper motor.

It reduces backlash almost completely, at the expense of adding the only two screws to the design.

mw0uzo, 2017-11-20:
Ah great to see the mods make it into the design :)
If you revive the old Z axis design with rails, I can test it. With the current design  I never got two steppers driven off one driver to work reliably, even at maximum current setting. So that might explains the trouble you were having. With separate drivers, i can put a significant amount of force on the Z axis and it will lift slowly without stalling. Any extra friction from rails won't be a problem and with the two Z endstop switches, jamming when misaligned will be less likely?
I'd be happy to test a revised Z axis!
If a new(/old) Z axis is forthcoming, I'll have to dismantle the machine so will try some screws in the gears.

xyzprints, 2017-11-22:
reverbat, can you post the ardunio editable files for marlin firmware? I'm currently trying to make it work with the v6 extruder with the drakon extruder (3:1 gear ratio extruder that is capable of working with multi materials including flexible materials) inside but, I can't make this work if I cant change the settings for the firmware which I am extremly new and bad at.

xyzprints, 2017-11-22:
I'll upload the design it after I make the firmware and confirm that it works

revarbat, 2017-11-23:
RAMPS1.4, with Marlin, with EEPROM support turned on (which I always do), lets you alter the steps per mm live with GCode similar to:

M92 X80 Y80 Z400 E1015 (set steps/mm for each axis)
M500 (store settings to EEPROM)

revarbat, 2017-11-23:
I've updated the RAMPS wiring diagram for the dual driver, dual Z limit switch design.  I'll work on updating the firmware on Friday.

revarbat, 2017-11-23:
Aaaaaand I fixed the image linking issues in the v2.0-Assembly.md wiki page.

revarbat, 2017-11-25:
I have pushed new firmware HEXfiles into the firmware folder, and have added a zip file of the Marlin sources configured for the Snappy without Heated Bed, and with the RepRapDiscount Full Graphic LCD.

revarbat, 2017-11-26:
Added insertion of set screws in for the drive gears.

revarbat, 2018-01-10:
Just checked in branch `v3.0` with the initial design for Snappy3 with substantially improved Z axis.

[image]

revarbat, 2018-01-10:
I'm marking this resolved as Snappy v3 should fix all the major issues listed.

closed 2018-01-10 by revarbat

mw0uzo, 2018-01-17:
Ah wow. Excellent!!!!!