revarbat/snappy-reprap #17: #extruder motor down  Firmware update & build volume change is it possible?

xyzprints, 2017-08-31:
I was trying out my snappy and my extruder motor isn't working. is there any possibility of finding the reason? and also, is there anny way of changing the build volume? last can I get the whole .ino file for this firmware? I would like to change some things in that firmware. and also, can I get some tips for the snappy?  like settings for cura.

mw0uzo, 2017-09-06:
Which firmware are you using? Pretty sure I had that problem when setting up Marlin, but now I'm using Repetier as it has backlash compensation - needed to get good circles. There was no backlash I could feel on my Snappy, but it was definitely needed to get good circles.

xyzprints, 2017-09-06:
Thanks for the reply!:)
I'm using the the (official) firmware. also, what is backlash?
I am using a ramps 1.4 and ardunio combo. I have a blow fan always on with 9x9 heat sinks.

mw0uzo, 2017-09-06:
Backlash occurs when the motor changes direction. If it is severe, you can feel it by wobbling the sled from side to side. e.g. if the gear and rack were heavily worn you would feel play in the mechanism. When the build is new, you can't feel anything, but there is a very small amount of backlash. Enough to affect circles!
So use the Repetier firmware I posted in the other thread. Marlin does not have backlash compensation.

mw0uzo, 2017-09-06:
There is a slight issue with the PID settings for the extruder temperature control in the firmware I posted. You might find the extruder temperature drifts up 10 degrees, I'm working on this problem so keep an eye on the temperatures and set it 10 degrees lower using the controller menu when it drifts up too much.

mw0uzo, 2017-09-06:
If you find settings that work well for you, let me know!

xyzprints, 2017-09-06:
Thanks for the reply! looks like that is some kind of wear and tear that I need to note with. I'll check if the firmware works without no problem after I get this z axis working. thanks for the warning! :)