revarbat/snappy-reprap #23: V3.0 extruder link

thannemann, 2018-03-01:
The V3.0 J-head hot end link to the amazon part is out of stock without an estimated replenishment date, is there a recommended replacement? Could this work with a Bowden extruder? I am not sure how the hot end mounts (just looking over the instructions) or will the Bowden be too big to mount

bjbeare, 2018-03-12:
Ditto here.

acorrow, 2018-03-13:
SainStore on eBay. Their listing for the 1.75 mm nozzle j head is slightly different than the model SS-110. I just picked one up and when it gets here I'll see if it fits perfectly and report back. Looks like it will, if anything is off will be the height and not by enough to make a difference I think.

http://r.ebay.com/S5oCA9

acorrow, 2018-03-14:
So heads up, the seller of that item from ebay contacted me to let me know their batch of these is broken and they can't sell them, and they have no more of them in stock, and they are discontinued :)

Sooooo yeah, gotta design a new one :) There's plenty out there that will do the trick, and for cheaper too I think. Just a matter of modding the openscad file. Likely want to update the BOM and stuff at some point, but still.

If i find more of these I'll let you all know.

revarbat, 2018-03-14:
I’ve designed up a new extruder platform for much more common E3Dv6 hot end, and checked into the v3.1 branch. I still need to finish the print cooling fan duct, though.

revarbat, 2018-03-15:
In the mean-time, this JHead extruder should work: http://a.co/eCIpayn

kwaad2, 2018-05-24:
@revarbat Hey, any chance you will be getting around to the part cooling fan for the E3D?
I've got most of my Snappy printed. (Just need the Z screws, and a few assorted random parts, and anything related to Extruder. (getting the E3D V6 3mm Volcano this weekend) Been holding off, as I expect to have to print it a few times to get the fit just right.

Also, any chance you can get the fan to blow from the "back" of the hot end? It's a system I don't think I've ever seen, but makes visibility so much nicer.

Depending on my luck, I will either run RAMPS with a Pi3 Klipper controlling it, or go full on, Duet Wifi.
I plan on having this bad puppy running at ERRF2018. :)

revarbat, 2019-02-23:
I've updated the links in the BOM.