revarbat/snappy-reprap #29: BOM links

Erhannis, 2018-11-02:
Quite a few amazon links in the V3.0 BOM have gone dead; it'd be nice if those got updated.  This may be some people's first printer they build, and they may not be confident they understand how to choose components.  Also, I note that the linked suggested power supply only goes up to 120W, so I'm not sure what to get if I have a heated bed and need 300W.

ManuGithubSteam, 2018-12-18:
Sadly wiki pages do not have pull requests...

Not even sure if the projet is still alive...

Anyway

Here are mine:

Qty	Description	Total Cost	Where Found
1	Power Supply, 120W. (300W w/ Heated Bed)	~$17	

1	Controller electronics. (RAMPS1.4+2560+Pololus)	~$30	
https://www.amazon.de/dp/B06XSZ9M77/?coliid=I27A2HBLU1ZE55&colid=1JQGPCRENC6K9&psc=0&ref_=lv_ov_lig_dp_it
3	Mechanical microswitch end-stops and wiring.	~$ 

5	NEMA17 Stepper Motors, 40mm length, 1m cable	~$58	
https://www.amazon.de/dp/B07DN97BGL/?coliid=I2MA092WJNIY8B&colid=1JQGPCRENC6K9&psc=0&ref_=lv_ov_lig_dp_it

1	e3dv6lite clone hot end w/heater, thermistor.	~$33	
https://www.amazon.de/dp/B018E6THV4/?coliid=I2MXHWAUPOAKNR&colid=1JQGPCRENC6K9&psc=0&ref_=lv_ov_lig_dp_it

2	40mm 12v 0.15A 2-pin cooling fan, 10mm thick.	~$16	
https://www.amazon.de/dp/B01CZFUOD0/?coliid=I21VE4U9NBKADG&colid=1JQGPCRENC6K9&psc=0&ref_=lv_ov_lig_dp_it

1m	Wire, dual conductor, 22 gauge, stranded	
https://www.amazon.de/dp/B00O7HPL5Q/?coliid=I27HGBPA3QF5KJ&colid=1JQGPCRENC6K9&psc=0&ref_=lv_ov_lig_dp_it

1	Heated Build Platform. (optional)	~$16	
Not so sure about this one, did not use one myself.
https://www.amazon.de/Homyl-3D-Drucker-Isolierung-Baumwolle-W%C3%A4rmed%C3%A4mmung-wie-beschrieben/dp/B07K31PJV4/ref=sr_1_14?s=diy&ie=UTF8&qid=1545150743&sr=1-14&keywords=3d+drucker+heizbett

You'll need the following Hardware to build the cartesian frame:

Qty	Description	Total Cost	Where Found
3.1Kg	Plastic Filament. (@ $13/Kg)	~$41	Amazon.com
1	Extruder Drive Gear	~$ 3	
https://www.amazon.de/dp/B06XBZ86Y3/?coliid=I2U35GP298WWX7&colid=1JQGPCRENC6K9&psc=0&ref_=lv_ov_lig_dp_it

1	686 bearing.	~$ 3
https://www.amazon.de/dp/B00K86FC86/?coliid=I39OKM2B719X82&colid=1JQGPCRENC6K9&psc=0&ref_=lv_ov_lig_dp_it

1	Borosilicate glass build platform. (214x200mm)	~$12	
https://www.amazon.de/dp/B07BM9MFKJ/?coliid=I3LIP0DSIRR3QB&colid=1JQGPCRENC6K9&psc=0&ref_=lv_ov_lig_dp_it

6	3mm x 12mm socket head screws.	~$ 1	Everywhere.
6	3mm nuts.	~$ 1	Everywhere.

ManuGithubSteam, 2019-01-27:
NOTE:

The motors are the wrong ! version!!!

revarbat, 2019-02-23:
I've updated the links in the BOM.

closed 2019-02-23 by revarbat