revarbat/snappy-reprap #31: Can't join lifter rods

brennancheung, 2018-12-22:
I broke 4 lifter rods trying to join them together.  I'm back to 0/4 lifter rods that I can use after a full day of printing.  There is not enough clearance to fit the rod into another rod.  I even tried sanding inside the hole and the fitting beforehand and adding lubricant as well.  I can only insert the piece about half way in before it snaps off.  A tremendous amount of pressure and rocking was needed to even get it that far.  Is there something I'm missing?

coyotebush2384, 2018-12-23:
The design I believe requires them to have a  solid fit I would just try sanding/filing there could be a lot of reasons they dont fit good probly your printer. Use the slope calibration to get parts dialed in. Mine took over 2 weeks and 3-4 printers running basically 24/7 I did build it so I had 2 ft of z height Haha but the point is..it works just have patience

brennancheung, 2018-12-26:
Might something with my settings then.  I still had to use a dead blow hammer to get the pieces in even with changing the dimensions.  I'll close the PR though since it seems like others have had success with the original size.

closed 2018-12-26 by brennancheung