revarbat/snappy-reprap #33: Firmware help

shade1059, 2018-12-24:
Are there instructions on how to flash the firmware? I am new to marlin. Specifically the LCD screen on the printer says "printer ready" but nothing else, and I do not know where to begin with troubleshooting.

ManuGithubSteam, 2019-01-06:
See here for example: https://www.my-home-fab.de/en/documentations/reprap-firmware/installing-marlin-firmware-on-ramps-1.4
Just leave out step 3 as the firmware is already modifierd for the snappy. So just dont do the modifications.

shade1059, 2019-01-10:
Thank you for the help. The Arduino software tells me that it compiled and flashed correctly, however the LCD screen comes up blank, and my computer fails to connect every time. Through some fiddling with the marlin settings, I can get the LCD screen to say "printer ready 16/0" but it is unresponsive beyond that.

ManuGithubSteam, 2019-01-10:
Hmm things that pop into mind:

- wirering issue
- you fried some part of the mega or ramps ( these boards are so cheap they
have no protection for " upps wire should go here not here)

- what fiddling do you do to get the LCD to work?
- did you try print with octoprint? Maybe LCD just borked

ManuGithubSteam, 2019-01-29:
Hi Shade 

i have good news.
I was able to compile the stock marlin firmware and get it to work on the ramps 1.4.
I uncommented the lines you mentioned. But i did NOT touch the motherboard line.
I strongly believe, you have to set it to Ramps SOMETHING (can't access config right now...)
Then i flashed it and the lcd did came on and worked with the buttons etc.
However i did not connect the motors and switches etc at that time.

Hardware i use: Mega2560 + Ramps 1.4 

Sadly i was not able to compile the snappy marlin Firmware. It gave me an error with the y Endstop.
Did you successful compile the snappy version of the firmware ?

If so, would you upload your sources here ?
Do you know if the snappy firmware does work with the Endstops switched from top to bottom ? (3.1 design)

As said the stock marlin does compile and flash onto the ramps. My Plan for now is to disect the config.h from the marlin and grad usefull stuff while hopefully let it compile and see where i go fro there...

Thanks for the help!

ManuGithubSteam, 2019-01-29:
Ok here it is. My first Marlin Firmware for snappy 3.1.

What did i do:
- basicly comb trough the orginal snappy config and copy all the sensible lines into the new Marlin config.h
- I did enabled a few nice things that are not in original snappy but i tried to keep it simple and not mess up all.
- I did not flip the z axis as it was switched back to the bottom of the z tower in 3.1 design.

It compiles.
So please test it. This is really alpha. I did not even test it on the device. just checked if it compiles!

Basically i think it will work but most likely needs some tinkering.
Please test.

And remember to set your correct motherboard in the config.h
[Marlin_Snappy_Next_v1.tar.gz](https://github.com/revarbat/snappy-reprap/files/2809341/Marlin_Snappy_Next_v1.tar.gz)

ManuGithubSteam, 2019-02-04:
Here is my semi final version.
- Axis go in the correct direction
- Endstops do work.
- Soft Endstops work

Over all this should print quite well. I have some issues with the z rods as they block if they are not 100% flat.
I did not test real printing yet as i have the above issue.

ManuGithubSteam, 2019-02-17:
Ok here is a working version with filament unload etc. I printed a benchy with it

[Marlin_Snappy_Next_v7.zip](https://github.com/revarbat/snappy-reprap/files/2873146/Marlin_Snappy_Next_v7.zip)

ManuGithubSteam, 2019-02-27:
Here is a new verision with curve cuttent to the steppers. Reduces vibrations quite a bit i think:
[Marlin_Snappy_Next_v8.zip](https://github.com/revarbat/snappy-reprap/files/2912245/Marlin_Snappy_Next_v8.zip)

shade1059, 2019-08-04:
Sorry for the delayed response. I was able to get it to work eventually. There were a few issues: The LCD shipped didnt exactly match what I ordered, and I didnt notice. Thus it didnt quite align with the model enabled in the firmware.
Also my power supply was having issues, I ended up getting an new one and all of the motors started working properly. Now I know I have the wrong number of steps per mm on my Z, but just gotta take some simple measurements to fix that (or see what you used in the marlin that you sent me).  I am good with marking this this issue taken care of.

closed 2019-08-04 by shade1059