revarbat/snappy-reprap #34: Motor resize doesn't seem right...

NigelThorne, 2019-01-13:
I have a bunch of NEMA 23 style motors that I was give... so I was trying to adjust the plans to with with those.

I replaced all calls to nema_motor_...(17) with nema_motor_...(23), which seems to have increased the space available to account for the bigger motors... however z_base_parts.stl  the legs intersect the motor housing... 
[image]

Is there something else I need to do? Or do I need to make changes in the plans to allow this change?
Cheers
Nigel
BTW... what a great idea for a project... Nice work! Well done for sharing what you have.

dorkmo, 2019-01-13:
Might just throw a difference() in there with nema cube

NigelThorne, 2019-01-14:
:) I fixed this part... I'll raise a pull request... but other parts also need adjusting... 

[image]

I think it will take widening a bunch of pieces... and maybe the motors will be to heavy... I plan to look into it.

revarbat, 2019-02-23:
Looks like you need to increase the `rail_height` in `config.scad` so that the motor fits in the X/Y motor segments.

revarbat, 2019-02-23:
Only two motors should matter for weight, as the Y and extruder motors are the only ones that move.