revarbat/snappy-reprap #36: BUG: Adjustment Screws -> Z sled needs other dimensions as build platform

ManuGithubSteam, 2019-01-27:
Hi

I printed a set of adjustment screws with my sloppy value and they are working fine in the build platform and are nice and tight but still usable.

However i could not risk using these in the z Rail connectors. They where to tight!

To get a good fit there i had to scale them a bit. 98 % shrinkage in x and y direction. Keep the z.

Then they fit into the z parts and work as expected.

revarbat, 2019-02-23:
I've added an explanation of using the Slop Calibrator to the wiki [v3.0-BOM](https://github.com/revarbat/snappy-reprap/wiki/v3.0-BOM) page, before the list of parts to print.

closed 2019-02-23 by revarbat

ManuGithubSteam, 2019-02-23:
Hi revbarbat,

Very nice to see you back on the wheel for this!

My sloppy value is 0.07. If there is such a thing as a too tight fit. Please mention it in the Slop Calibrator description :-)

revarbat, 2019-02-24:
In a perfect world, with an ideal 3D printer that printed to exact dimensions, printer_slop would be 0.00.  I've yet to see said perfect printer, though.

revarbat, 2019-02-24:
Though I guess a too-tight fit is one where the plug takes more than light-force to insert and remove in any rotation.

ManuGithubSteam, 2019-02-24:
Hello revarbat,

Well the prusa mk3 has a sloppy value of 0.07 for me. But I need quite a
bit of force to remove the parts once connected. Hence my question if it is
possible to have a too tight fit and have adverse effects like the X
vibration..

Thanks