revarbat/snappy-reprap #37: Sliding of Y sled partial assembly on Y axis rail assembly

parth1739, 2019-02-14:
Hi,
Y sled assembly is not sliding smoothly over Y axis rail assembly. I have tried lubricating it with vaseline and mineral oil but t doesn't help. I have also tried filing the parts but didn't help. 
Also, I faced problems connecting all the parts. I had file them for hours in order to fix them together and still the results are not good. Has everyone faced similar situations?

coyotebush2384, 2019-02-14:
Yes that's why you would want to use the sloop calibrater and openscad to get parts to print correctly

parth1739, 2019-02-14:
Hey,
Thanks for the prompt reply. I don't know what sloop calibrater is. And I am not familiar with Openscad either. Can you please guide me?

coyotebush2384, 2019-02-14:
I probably can. hopefully someone else will jump in but if not I can give better reply when I get off work

dorkmo, 2019-02-14:
in config.scad look for printer_slop

dorkmo, 2019-02-14:
i saw on forum people talking about this file https://github.com/revarbat/snappy-reprap/blob/master/STLs/slop_calibrator_parts.stl

parth1739, 2019-02-15:
> in config.scad look for printer_slop

Hey Dorkmo,
Thanks for the reply. I opened config.scad and found this -> "printer_slop = 0.20;  // mm". Now I understand that slop callibrater is something which is used to fit parts correctly, but I cannot figure out how do I use it? Also, I have already printed slop callibrator parts.

dorkmo, 2019-02-15:
I havent built a snappy, but i think after you print the slop_calibrator_parts.stl, you trying putting the peg into the holes, after you fiigure out the best fit, you read the number off of the side and define that number in the printer_slop= value. Then that value is used to generate the models for the rest of the snappy. Have you tried placing the peg in the holes? Let us know what it fits in.

dorkmo, 2019-02-15:
On a side note, i couldnt find any reference to the calibrator_parts in the build guide. i think it probably needs a pull request.

parth1739, 2019-02-16:
I did not try it earlier. But now it fits perfectly in "45" and the fit it so strong that the peg is not coming out anymore.

dorkmo, 2019-02-16:
cool. I think that means you should set printer_slop=45; then run the make file

parth1739, 2019-02-17:
Alright. The change in printer_slop value will be done in the configuration file that I have in my system right? The value that is mentioned there right now is ".20". So should I mention the new value as ".45"? How do I run this file? 
Also, I have printed almost all the parts and it has taken me a long time. I was hoping there is some way to utilize those parts as well.

dorkmo, 2019-02-17:
I think you might be able to just reprint the parts that are giving you the most trouble? besides the Y axis, how bad off are the other parts? are they wobbly or something else?

After you change the slop value in the config file on your computer, you'll need to use a terminal to run the make command. That executes all the stuff in https://github.com/revarbat/snappy-reprap/blob/master/Makefile which will create custom STL files using the new slop value. At least i think thats my understanding of the readme explination. hope it works out for ya.

parth1739, 2019-02-17:
The parts like sled and motor assembly needs to be smooth in order to slide properly. I think they need a reprint. Also, I was able to fit parts like rail segments and motor assembly after filing but now they are  bouncing back once I join them, bad fit.  Since I got stuck in the initial stage I still got to check the rest of the parts, But I am certain that they would require filing too.  
Where will I find the make command? The link  you shared, is that the make command? Also, since I use windows its a bit different for me. I think I have to carry out the modifications for every individual parts. I read something like that in readme file.

dorkmo, 2019-02-17:
Yeah maybe it will be easier to do them individually, just open each file scad file, then press F6 to render it, then go to File > Export > as STL

parth1739, 2019-02-17:
Alright. I will just tell you the process which I followed. Correct me if I went wrong somewhere. Firstly, I changes the slop value in config file. I saved that file. Then I open the design file that I need to make as per the correct dimension and then render it using F6 and then export as stl. 
Is that correct or am I missing any step?

dorkmo, 2019-02-17:
yup that should be right. like if you look at https://github.com/revarbat/snappy-reprap/blob/master/xy_sled_parts.scad for example, the first line shows "include <config.scad>", then it uses the slop value from that

parth1739, 2019-02-17:
Great thanks a lot. I am a bit curious about slop calibrater. Was it particularly designed to calibrate the parts as per the user's printer? Is it used in for some other applications also?

dorkmo, 2019-02-17:
i've been doing a lot of guessing, but i think thats its only application.

ManuGithubSteam, 2019-02-17:
After yout put the correct slop value just type make in the main directory and all files will be rerenderd with the corret slop in the STL folder.

parth1739, 2019-02-18:
Thanks @ManuGithubSteam . Can you tell me what do you mean by main directory and where can I find it?
@dorkmo  I came across another problem today. I tried putting the peg of slop calibrater in other holes and found that it is fit for other holes also. Like its going easily in 45, 40, 35 etc.. What do I do in this situation?

parth1739, 2019-02-20:
Hi,
When I was rendering the Y_z joiner parts I received a warning stating "Object may not be a valid 2-manifold and may need repair!"

ManuGithubSteam, 2019-02-20:
Hi

The main directory is the directory, that you download from githup or the
one that gets created when you clone the repo. It has a subfolder STL and a
file called Make in it.

Slic3r might be able to repair your broken STL file.

Hope that helps

revarbat, 2019-02-22:
Yes, that should be 0.45, not 45.
Also, 0.45 is a very large printer_slop value.  Are you sure your current printer is printing square and isn't wobbling on the Z axis over much?

- Revar

revarbat, 2019-02-23:
I've added an explanation of using the Slop Calibrator to the wiki [v3.0-BOM](https://github.com/revarbat/snappy-reprap/wiki/v3.0-BOM) page, before the list of parts to print.

closed 2019-02-23 by revarbat

parth1739, 2019-02-23:
> Yes, that should be 0.45, not 45. Also, 0.45 is a very large printer_slop value. Are you sure your current printer is printing square and isn't wobbling on the Z axis over much? - Revar
> […](#)

Hey @revarbat 
Earlier the peg was not a good fit with any hole except 0.45. But now I guess with smoothing due to push in and pull out it is fitting in other holes aswell.  I currently have 2 printers. As per your new instructions on github page I think now slop value for one of my printer in .40 and the other is .35. 
Talking about wobbling  of z-axis of current printer. It does vibrates a bit while printing but I haven't confirmed whether its printing exact square or not. Any added instructions from your end could be useful for me. 

Thanks

parth1739, 2019-02-23:
> Hi The main directory is the directory, that you download from githup or the one that gets created when you clone the repo. It has a subfolder STL and a file called Make in it. Slic3r might be able to repair your broken STL file. Hope that helps
> […](#)

Hi @ManuGithubSteam 
Alright now I understand what the main directory is. I also found the makefile in it. Could tell me where in the main directory should I  type "make" to render files in STL folder.