revarbat/snappy-reprap #39: X axis movement -> very high vibrations ruin print, Y Axis ok

ManuGithubSteam, 2019-02-17:
Hi Guys,

I just finished my snappy 3.1 and its an amazing machine. However, i have some bad vibrations when the x axis is moving. So bad in fact, that it ruins the print. Moves in y axis are fine and they look really good. But all the moves in the x direction do vibrate so much, that there are artifacts in the print and that constantly. It just vibrates too much.

Any ideas why the x axis is vibrating that much ? Could this be that there is not enough current on the x stepper motor? 

Thanks

dorkmo, 2019-02-18:
you might upload a video to youtube

ManuGithubSteam, 2019-02-18:
There you go: https://youtu.be/dKV7vusrJ4g
Hope it helps
Any ideas ?
Thanks!

revarbat, 2019-02-23:
Try lubricating the sliders and/or the gear racks with vasoline.  Don't know if that'll help or not.  I'm seeing some excess vibration on my own X axis, and I'm worried that it may be because the drive motor is working against more of a lever arm at the ends of the sled racks.

ManuGithubSteam, 2019-02-23:
Hi revarbat,

Nice to see you back! I'm building this snappy for a school cooperation and it would be super if we could get to the bottom of this :-)

I tired so far:

- Increase current on the x axis motor = no effect
- reduce jerk settings to 2 = could have helped a bit not sure
- decrease x motor acceleration to 200 = This helped 

The problem with the decrease of the acceleration i fear is: The problem gets better to to totally avoid it i would have to decrease the acceleration probably to 50 or even less. 

Is there a minimum amount of acceleration ? What does this mean print time wise ?

Also when i reduces the jerk settings, the print time got way longer.
 
I saw a firmware setting, to immediately accelerate to motors could that help ?

Speaking of firmware: I tried to model my version after your settings, but i'm not sure if i have the best configuration. If you want you can use my version as a base for 3.1 firmware :-)

Here are some photos from a benchy with 200 x accel:

[image]
[image]
[image]
[image]

revarbat, 2019-02-24:
When I slowed my acceleration, it mostly just made the wavelength of the ripples longer.  While my prints show the ripples at the ends of longer X motions, your images seem to show ripples at the middle of the print.  I don't feel vibration as I manually slide my X axis by hand, but I can see it when printing.  Perhaps it's because the drive gear pushes the rack (and thereby sled) towards just one side?

ManuGithubSteam, 2019-02-24:
Hi

Yes i have a similar experience. When i use marlin to move the axis it is very smooth without vibration i believe. But when i print gcode there is vibration. When i print, it seems that the vibration only occurs in the direction to the right but not on the back move. However the benchy has the ripples on both sides. So it is present also on the back move.

I now use x acceleration of 100 and the ripples are much less but still there. What acceleration do you use ? Could this issue arise because of incorrect stepps at the motor ?

ManuGithubSteam, 2019-02-24:
Ok 80 mm/sec x accel and lithium grease on the rails seem to have nearly fixed the issue for me. I think i will go down to 70 mm/sec to make them even less visible but the downside is, that the print time nearly doubles. But i'm glad i can produce decent prints.

I will report back in detail with a picture, when the print has finished...

Here is my config.

[Slic3r_config_bundle.ini.txt](https://github.com/revarbat/snappy-reprap/files/2898237/Slic3r_config_bundle.ini.txt)

ManuGithubSteam, 2019-02-28:
Ok so now im printing with 65 mm/sec x axis acceleration and the vibrations are still there but only visible if you look directly for them. So the issue remains but this is a workaround i would suggest.
Also i enabled curved acceleration in the firmware but that does not help so much only on small parts it makes a difference...

ManuGithubSteam, 2019-03-24:
Hi @revarbat 
Ok i have more info on this issue:

1. I mitigated the vibrations as i wrote above by reducing the a axis acceleration down to 60mm/s (from 400) i also set the feedrate to 150 mm/sec. I did this in the speed settings of the "print" profile (Accerlation for perimeters 60) and as a setting in the Printer profile (Machine Limits). In the Firmware i left it on 400.

This helped to make the vibrations barely visible. But they are still there. 

 2. After that i switched from A9688 to TMC2208 in an effort to make the printer more silent. I switched out X, Y and Z stepper drivers. This worked like a charm and the printer is now super silent.

However as the TMC2208 does advertise speed gain while maintaining print quality i tried some faster speeds. In the end i can say that even with 400mm/s acceleration the loss in quality is like 10% to the quality when printing with 65mm/sec speed.

So the drivers helped to get rid of the excess print time. For me this is a huge win. However it got me thinking. If the TMC can print with less vibrations, doesn't this mean that the vibration is caused by "uneven" sinus signal curve of voltage, as the TMC does essentially smooth out the uneven current into a smooth more sinus curve like curve. 

If this is the case could we get rid of the vibrations nearly all together if we dial in the voltage exactly on the amount that is needed and just enough to no skip steps ?

Sadly i could not try this out as i had to deliver my snappy to the school and i had to eyeball the amount fast.

Other ideas i had to maybe solve this:

- Make the y axis heavier and so not so much vibrations transfer might occur ?
- Make a second line of fish grip rails for the gears so the gear is gripped from two sides not just one
- Make a padding of some sort for the x motor, That not so much vibration is transfered to the rails via direct contact with the motor. For other cartesian printers there a special dampeners for this.
--> It could be enough to not let the motor contact on the top of the rails and only have a hold further down and on the 4 corners ?

I hope we find a solution before they finish printing their snappy on mine :-)

PhilMegman, 2020-05-12:
Hi everyone,

did you concider the "stick-slip phenomenon" as reason for your issue? https://en.wikipedia.org/wiki/Stick-slip_phenomenon

My snappy has a different issue, but I think it is caused by the stick-slip phenomenon. (It is not performing small movements in X-axis.)

So if anyone knows a solution for this I would like to hear it.

Best regards

revarbat, 2020-05-12:
I'm not exactly an engineer.  My best guess is that we need a better lubricant than mineral oil or vaseline. My experience with various lubricants is as follows:
- Vegetable oil worked, but will eventually go rancid.
- Mineral oil works, but I still get vibration in my builds. I also have to re-apply it periodically.
- Vaseline doesn't need to be re-applied as often, but it has more friction, and I still get vibration in my builds.
- 3-in-1 oil slowly soaks into the PLA of the rails, expands them, and makes them bind.
- Spray-on dry PTFE lube actually made the rails have *more* friction.

I gather that many petroleum based lubricants may degrade plastics over time anyways.

Some things to try:
- Graphite dry lubricant
- ???

ManuGithubSteam, 2020-05-12:
Hi guys,

Just to chip in; I tried graphite based lubricant on the X axis and it
helped but not so much that I could actually print with the snappy. For now
this project is dormant for me.

However if the version gets fixed I will gladly resume it.

I also have some designed parts that where missing or did not fit with my
motor's. If interested I can share them with you.

revarbat, 2020-05-12:
I'se seen recommendations for white lithium grease and silicone based lubricants for use with PLA and ABS.

revarbat, 2020-05-12:
Regarding having herringbone racks on both sides of the drive gears, that has the problem that the two sides of gear are moving in opposite directions.

I have been thinking, though, about making a screw drive for X and Y similar to the Z axis lifter rods.  That wouldn't have the issues that the herringbone rack has on applying pressure to only one side.  May have more issues with backlash, though.  Also may limit the speed of the X and Y axes.

dorkmo, 2020-05-13:
ive used this in other projects, its okay but still not a magic bullet.
https://www.super-lube.com/silicone-lubricating-grease-with-syncolon-ptfe

PhilMegman, 2020-05-13:
> I'm not exactly an engineer. My best guess is that we need a better lubricant than mineral oil or vaseline.  [...]
> 

I've tested different lubricants too and all of them change the behavior while the sled moves, but none of them changed the behavior from start up.

Simple test: Take only the three rail parts and the sled. Both clean of grease. Make them a "Inclined plane" and see the angle when the sled starts moving. Repeat the test with grease. Don't shake it ot touch the sled. In my tests the angle was always nearly the same. With some grease it get's worst. But none of the grease caused a smaller angle (less static friction).

I've glued the three parts of each rail together to get more stiffness. In addition I've re-printed the Y-sled with smaller slop. After this, I had no vibration any more. I still have ghosting, but no vibrations visible on the perimeters.

BUT now I have a different problem, that was maybe hided by the vibrations before: 
As mentioned in my post before, my snappy does not make small movements in X-direction. And for longer moves it just stops before a corner instead of moving slow. That leads to a calibration cube (20mm x20mm x20mm) with X:19.3mm Y:20.4mm Z:20.4mm.

Grease as mentione before doesn't helped
Higher current on the steppers does not lead to a solution, but melts the drive gear, finally.

Maybe we need to think about smaller touching area of the sled to the rail...
I think I will try a different material for sled or rail next. Maybe PLA on PLA have a too high static friction. Maybe I try one part in PETG or I try to get metal in between the two PLA parts.

M15631563b, 2024-07-04:
> من دقیقا مهندس نیستم بهترین حدس من این است که ما به روان کننده بهتری نسبت به روغن معدنی یا وازلین نیاز داریم. تجربه من با روان کننده های مختلف به شرح زیر است:
> 
> * روغن نباتی موثر بود، اما در نهایت فاسد می شود.
> * روغن معدنی کار می کند، اما من هنوز در ساختمان هایم لرزش دارم. همچنین باید به صورت دوره ای آن را دوباره اعمال کنم.
> * وازلین نیازی به بارگیری مجدد ندارد، اما اصطکاک بیشتری دارد و من همچنان در ساختمان هایم لرزش دارم.
> * روغن 3 در 1 به آرامی در PLA ریل ها فرو می رود، آنها را منبسط می کند و باعث اتصال آنها می شود.
> * روغن PTFE خشک اسپری شده در واقع باعث می شود ریل ها اصطکاک _بیشتری_ داشته باشند .
> 
> به نظر من بسیاری از روان کننده های نفتی ممکن است به مرور زمان پلاستیک ها را تخریب کنند.
> 
> برخی از چیزهایی که باید امتحان کنید:
> 
> * روان کننده خشک گرافیت
> * ؟؟؟

If you don't have a hot container, try soap

M15631563b, 2024-07-04:
If the hot container hardens, it will not damage the soap.  Try vegetable soap or margarine

>

M15631563b, 2024-07-04:
I doubt that lubrication can solve your problem.  Check the bearing of the stepper motor, especially the spring at the bottom of the shaft