revarbat/snappy-reprap #40: Filament is escaping the extruder idler

ManuGithubSteam, 2019-02-28:
As you can see the filament is escaping the idler to the side. This leads to the phenomena that there is no more extrusion taking place and the print is ruined.

I made a fix for it here but wanted to raise the issue to fix it at the source. (I cant do openscad so i fixed it in tinkercad): https://www.thingiverse.com/thing:3459378
[image]
[image]

revarbat, 2019-03-23:
Ah. Your drive gear doesn't have the self-centering groove that the mk8 drive gear has.
https://folgertech.com/products/mk8-drive-gear-for-reprap-extruder-5mm-bore-and-7mm-effective-dia-mk-8

ManuGithubSteam, 2019-03-23:
Hi Revar,

I was under the impression, that the groves in that gear would off center
the filament path into the extruder. And did not use it. Is this the case?
I just did a visual check...

Have a nice day
Manuel

revarbat, 2019-03-23:
You can just re-center the drive gear groove over your filament path by re-adjusting where it is on the motor shaft.  Up to and including turning it around backwards.  Shouldn't need to, though.  In any case, as you've found out the hard way, the groove keeps it in line.

- Revar

ManuGithubSteam, 2019-03-24:
Hello Revar,

I know but I mean not in the X direction where I can slide the gear into
position. I mean in the y direction. So to speak that the groved gear
"hole" (seen from top) is not centerd on the hole below because of the
groved out material. As I see it with the grove it is not a straight line
anymore. But I could be wrong.

Also: in the assembly manual the gear looks more like mine and not a groved
one.
Manu