revarbat/snappy-reprap #43: Can not get makefile to exicute

mlpage34, 2019-03-09:
I have installed lmde3 and installed openscad, after changing the makefile, I try to run make, and get a syntax error on the first line <!DOCTYPE html>. 
I have tried my google fu and it is not strong. can someone point me in a direction to go in order to find a solution?

revarbat, 2019-03-09:
What error does it give?

mlpage34, 2019-03-09:
Makefile:7: *** missing separator.  Stop.

ManuGithubSteam, 2019-03-10:
What system do you use ? Linux or Windows or Mac ? The text editor you use must be capable of using the right end of line (Unix) and you need to save it in unix format.

mlpage34, 2019-03-10:
that was using linux. I can install a different distribution, thats not a problem. I have ubuntu, xubuntu, lmde, kali, debian 9, kubuntu, and lubuntu. also, I can use a diffrent editor. I used nano to modify it last time. It was my intention to use LMDE 3 only for 3d printing, but if I would have better support with fedora, debian, or ubuntu, I would swich

ManuGithubSteam, 2019-03-10:
No thats all fine. So its a logical error in the file, Made by you and no machine can exacly say what it is but its on line 7.
best idea: See the orginal file and compare line 7. It seems some seperator is missing. It seems like you should not modiy that line (7). I did not when i made mine.

I attached mine. It works.

[Makefile.txt](https://github.com/revarbat/snappy-reprap/files/2949713/Makefile.txt)

ManuGithubSteam, 2019-03-20:
could you solve it ?

mlpage34, 2019-03-21:
Yes, it was because I didnt clone the repository. the makefile was full of html and xml language. once I cloned, everything was good

closed 2019-03-21 by mlpage34