revarbat/snappy-reprap #44: Request: less vitamins,(with solution)

ManuGithubSteam, 2019-03-16:
Hi

I'm looking into ways to get less vitamins for the snappy.

What you can do:

- for the y gears part: Use gyroid infill 75 % and make it tight! and use expo glue to glue the shaft in. 
This works if you file the rails down with 400,600,800 and 100 grit sandpaper. 
Then we safe 1 screw and one nut :-)

WARNING: You NEED a screw for the x axis gears.!!! I tried it! The x axis is much heavier and needs to be secured with a screw.

To test:

- Screws on the z axis.

revarbat, 2019-03-23:
Iiinteresting.  I originally tried press fit, but I ended up with the shaft loosening, even keyed.  So I put in the screws.  I'd already had a place to put the screws, luckily.  I didn't really get around to trying high-density or alternate infills.  The new slic3r gyroid infills look really interesting.

ManuGithubSteam, 2019-03-23:
Hi ;)

Yes but it only worked on the y rail for me. The X is just to heavy.

But I can imagine that petg with gyroid could hold as it has more give and
is not so brittle. This would need testing but my snappy is in production
already.

Thanks

ManuGithubSteam, 2019-08-10:
Ok did some more tests. It does not work. You need the screw in the y and x asis.

closed 2019-08-10 by ManuGithubSteam