revarbat/snappy-reprap #49: Make errors.

anastazya, 2020-02-08:
./stl_normalize.py -c STLs/bridge_segment_parts.stl -o STLs/bridge_segment_parts.stl
/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD -m make -o STLs/cable_chain_link_parts.stl cable_chain_link_parts.scad
WARNING: rx was assigned on line 402 of "GDMUtils.scad" but was overwritten on line 404
WARNING: ry was assigned on line 403 of "GDMUtils.scad" but was overwritten on line 405
WARNING: variable center not specified as parameter, in file cable_chain.scad, line 16
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: variable center not specified as parameter, in file cable_chain.scad, line 16
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: variable center not specified as parameter, in file cable_chain.scad, line 16
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: variable center not specified as parameter, in file cable_chain.scad, line 16
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: variable center not specified as parameter, in file cable_chain.scad, line 16
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: variable center not specified as parameter, in file cable_chain.scad, line 16
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: variable center not specified as parameter, in file cable_chain.scad, line 16
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: variable center not specified as parameter, in file cable_chain.scad, line 16
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: variable center not specified as parameter, in file cable_chain.scad, line 16
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: variable center not specified as parameter, in file cable_chain.scad, line 16
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: variable center not specified as parameter, in file cable_chain.scad, line 16
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: variable center not specified as parameter, in file cable_chain.scad, line 16
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: variable center not specified as parameter, in file cable_chain.scad, line 16
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: Too many unnamed arguments supplied, in file GDMUtils.scad, line 857
WARNING: variable center not specified as parameter, in file cable_chain.scad, line 16
[... 62 lines cut ...]
WARNING: ry was assigned on line 403 of "GDMUtils.scad" but was overwritten on line 405
WARNING: Ignoring unknown variable '$do_prerender', in file GDMUtils.scad, line 7.
WARNING: variable center not specified as parameter, in file GDMUtils.scad, line 914
WARNING: variable center not specified as parameter, in file GDMUtils.scad, line 915
WARNING: variable center not specified as parameter, in file GDMUtils.scad, line 918
WARNING: variable center not specified as parameter, in file GDMUtils.scad, line 919
./stl_normalize.py -c STLs/cooling_fan_shroud_parts.stl -o STLs/cooling_fan_shroud_parts.stl
/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD -m make -o STLs/drive_gear_parts.stl drive_gear_parts.scad
WARNING: rx was assigned on line 402 of "GDMUtils.scad" but was overwritten on line 404
WARNING: ry was assigned on line 403 of "GDMUtils.scad" but was overwritten on line 405
WARNING: Ignoring unknown variable '$do_prerender', in file GDMUtils.scad, line 7.
./stl_normalize.py -c STLs/drive_gear_parts.stl -o STLs/drive_gear_parts.stl
NON-MANIFOLD DUPLICATE FACE! STLs/drive_gear_parts.stl: [4.585, 10.733, 10.45] - [5.279, 10.408, 9] - [5.28, 10.408, 9]
NON-MANIFOLD DUPLICATE FACE! STLs/drive_gear_parts.stl: [-4.585, -10.733, 10.45] - [-5.279, -10.408, 9] - [-5.28, -10.408, 9]
NON-MANIFOLD DUPLICATE FACE! STLs/drive_gear_parts.stl: [-10.733, 4.585, 10.45] - [-10.408, 5.279, 9] - [-10.408, 5.28, 9]
NON-MANIFOLD DUPLICATE FACE! STLs/drive_gear_parts.stl: [10.733, -4.585, 10.45] - [10.408, -5.279, 9] - [10.408, -5.28, 9]
NON-MANIFOLD DUPLICATE EDGE! STLs/drive_gear_parts.stl: [10.408, -5.279, 9] - [10.408, -5.28, 9]
NON-MANIFOLD DUPLICATE EDGE! STLs/drive_gear_parts.stl: [-4.585, -10.733, 10.45] - [-5.279, -10.408, 9]
NON-MANIFOLD DUPLICATE EDGE! STLs/drive_gear_parts.stl: [4.585, 10.733, 10.45] - [5.279, 10.408, 9]
NON-MANIFOLD DUPLICATE EDGE! STLs/drive_gear_parts.stl: [5.279, 10.408, 9] - [5.28, 10.408, 9]
NON-MANIFOLD DUPLICATE EDGE! STLs/drive_gear_parts.stl: [10.733, -4.585, 10.45] - [10.408, -5.279, 9]
NON-MANIFOLD DUPLICATE EDGE! STLs/drive_gear_parts.stl: [-5.279, -10.408, 9] - [-5.28, -10.408, 9]
NON-MANIFOLD DUPLICATE EDGE! STLs/drive_gear_parts.stl: [-10.408, 5.279, 9] - [-10.408, 5.28, 9]
NON-MANIFOLD DUPLICATE EDGE! STLs/drive_gear_parts.stl: [-10.733, 4.585, 10.45] - [-10.408, 5.279, 9]
make: *** [STLs/drive_gear_parts.stl] Error 255

buddybu, 2020-06-04:
I am seeing this same error.  Help?

dsbaars, 2020-08-26:
Same here, compiling manually in the OpenSCAD gui does work. It is the "stl_normalize.py" script that generates errors. I tried commenting out the drive_gear_parts and xy_sled_parts to make the rest compile, since I only need the glass and heated bed clips changed and those throw errors but somehow that does not fix it.